thehookup/Motorized_MQTT_Blinds #3: Battery

danpowell88, 2019-04-21:
Any chance you could do an update or show an example of the type of battery that would be compatible with this instead of requiring AC power

recklessnl, 2019-06-05:
Seconding this, some locations just have no way of wiring it to AC power without it looking awful.

Additionally, I saw someone using the solar power strips strapped on the glass, which you couldn't see from inside, but they would charge the battery during the day. Is this something that's feasible with this build?

thehookup, 2019-09-21:
I have a working prototype of this, trying to ensure its safety and longevity before I release it into the wild.

tung256, 2019-11-03:
that would be awesome!
i volunteer to be the first to try if the battery can last at least 2 weeks.

incognit-joe, 2019-11-13:
Hi, I have built the kit and have a dual motor set up. Everything running perfectly thanks! I would also be interested in the battery option. Can you advise if you've had any progress with this as i would be pretty gutted if i proceed to drill holes in my walls then a week later you put up a tutorial to add a battery..

Either way i want to extend my thanks, this is the first electronic project i have ever tried and im over the moon to have succeeded!

Rgds,
Joe

thehookup, 2019-11-13:
In my initial test the battery only lasted 2 months, even with the solar panels on it.  I need to revamp the code a bit more but currently have no time to devote to it.  Probably in mid-december I'll revisit.

joelwetzel, 2020-02-21:
Any updates?  I've built 2 of these awesome things now.  But would love to get rid of more cables.

1ntroduc3, 2020-07-26:
> In my initial test the battery only lasted 2 months, even with the solar panels on it. I need to revamp the code a bit more but currently have no time to devote to it. Probably in mid-december I'll revisit.

Do you have any update?

JustMaier, 2020-09-09:
@thehookup Any chance you can at least share a few of the details on what components you used and maybe what adjustments you made? 2 months on battery is actually pretty good.

I'm going to attempt the battery-powered route here's my plan.

**Additional Components:**
- 5000mAh USB battery pack
- [7 Solar Cells (60mA)](https://smile.amazon.com/gp/product/B0736W4HK1) + [Schottky Diodes](https://smile.amazon.com/gp/product/B079KDQQPD) (to bypass dark cells) (anticipating 70mA - 210mA total)
- Micro USB cable to recharge battery pack from 5V from solar cells
- [Step Up Converter (5v to 12v)](https://smile.amazon.com/gp/product/B07RNBJK5F) between the battery pack and driver (since my base power will be 5V, I plan on dropping the buck converter)

**Code Adjustments:**
- [Modem sleep](https://diyi0t.com/how-to-reduce-the-esp8266-power-consumption) the device and wake it every 5 seconds to check for MQTT messages. _Optimally_ reduces daily power consumption by ~60% (80% of every minute it's running at 25mA instead of 100mA)
- Auto close and deep sleep from 8pm - 8am. _Hopefully_ reduces daily power consumption by an additional ~37% (I believe it still consumes 10mA in this mode)
- Assign static IP (reduces the power needed on wake)

**Hopeful Outcome:**
- - 600mAh/day for nodemcu
- -5mAh/day for motor (assuming 1.25mAh per open or close with 4 operations per day)
- +560mAh on bad days or +1,680mAh on good days from solar cells

**Potential Problems:**
- Solar is likely to underperform what I've hoped for. I tried to measure output with a multimeter on my desired windows and a single cell was producing 3mA at noon on a sunny day which is considerably less than even my "bad day" estimate. Maybe different panels would help, or maybe my window just doesn't get enough light... Either way, if I'm only able to get 168mAh/day from the cells, I can only expect to go 10 days or so without having to recharge
- I haven't accurately measured the consumption of the actual motors to open and close the blinds. I figure they'll only be used 2-4 times a day and I'm hoping that I can essentially just not worry about their overall power draw with such limited use. That assumption could be very wrong.
- Sleep cycling the nodemcu like that might not actually save as much energy as I hope
- I'll probably have to tweak the USB battery pack so that it doesn't turn off when the nodemcu deep sleeps
- I've never done something like this before and I could be horribly horribly wrong with all my calculations and components needed 😂

Mehick57, 2021-05-04:
> @thehookup Any chance you can at least share a few of the details on what components you used and maybe what adjustments you made? 2 months on battery is actually pretty good.
>
> I'm going to attempt [...]

Sounds like a great idea and plan! I am just wondering if you have gotten this to work? This is exactly what I am looking for since I can't run wires to ten sets of blinds I would like to convert! Thanks for your time!

Mehick57, 2021-05-04:
> In my initial test the battery only lasted 2 months, even with the solar panels on it. I need to revamp the code a bit more but currently have no time to devote to it. Probably in mid-december I'll revisit.

It would be great if one of you geniuses could figure this one out!

JustMaier, 2021-05-04:
> > In my initial test the battery only lasted 2 months, even with the solar panels on it. I need to revamp the code a bit more but currently have no time to devote to it. Probably in mid-december I'll revisit.
> 
> It would be great if one of you geniuses could figure this one out!

I actually abandoned this route. In my findings, it seemed that WiFi takes too much energy to be able to run on a battery for an extended period of time.

The best way to do this in my research was actually using RF on a micro-controller designed for low-power usage, like a moteino. This was a [helpful guide project](https://snorp.net/2016/05/20/blinds-controller.html).

Mehick57, 2021-05-05:
Thanks so much for your
          quick reply and being pointed towards this project! I will be
          trying to see if I can put this together and get it working on
          HA!
    
    On 5/4/2021 4:54 PM, Justin Maier
      wrote:
    
    
      
      
        
          In my initial test the battery only lasted 2 months, even
            with the solar panels on it. I need to revamp the code a bit
            more but currently have no time to devote to it. Probably in
            mid-december I'll revisit.
        
        It would be great if one of you geniuses could figure this
          one out!
      
      I actually abandoned this route. In my findings, it seemed that
        WiFi takes too much energy to be able to run on a battery for an
        extended period of time.
      The best way to do this in my research was actually using RF on
        a micro-controller designed for low-power usage, like a moteino.
        This was a helpful guide project.
      —
      [
{
"target": "https://github.com/thehookup/Motorized_MQTT_Blinds/issues/3#issuecomment-832275991",
"url": "https://github.com/thehookup/Motorized_MQTT_Blinds/issues/3#issuecomment-832275991",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]