thehookup/Motorized_MQTT_Blinds #5: Smartthings integration

mbecraft76, 2019-05-06:
Flashed your bin file for the Smartthings integration. Flash went fine on a brand new nodemcu board. Parent device created properly as I've created a dozen or so with the ST Anything parent/child devices. The nodemcu board seems to connect to smartthings just fine, Refresh works just fine. It shows up in the Smartthings log when refreshed. But no child devices have been created. Which ST Anything child device is this supposed to take advantage of? The available ones from Dan/Daniel don't seem to support the stepper motors.

CurtisFeatures, 2019-09-16:
> Flashed your bin file for the Smartthings integration. Flash went fine on a brand new nodemcu board. Parent device created properly as I've created a dozen or so with the ST Anything parent/child [...]

@mbecraft76 did you manage to get this going? I’ve also been unable to get it working today 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

mbecraft76, 2019-09-16:
@CurtisFeatures  no I never got it going. I posted here and posted on Reddit where I first saw this project. I may have even posted on youtube about it, I don't remember. All I got was crickets. I'm not sure if this guy even cares about the smartthings integration. Dude has some decent ideas, but followup and finesse need a lot more work. All I know is that I wasted my money on parts for this and ended up using the ST Anything servo libraries and a few servos instead. I also 3D printed adapters for the servos to fit in the blinds. They were cheaper and easier to use anyway. No janky process of cutting traces like on the stepper motors or adding buck converters or driver boards. I'm sure the stepper motors are quieter, but I really don't care about the noise. If you'd like, I can point you to the posts/threads on smartthings that I used to make mine work. If someone ever comes up with a reliable stepper motor library that integrates with ST Anything, I may swap over to them just to compare.

CurtisFeatures, 2019-09-16:
> @CurtisFeatures  no I never got it going. I posted here and posted on Reddit where I first saw this project. I may have even posted on youtube about it, I don't remember. All I got was crickets. I'm [...]

@mbecraft76 I got it working with MQTT and openHAB for home use and it’s been working fine. I also made my own 3D printed enclosure as I use it with virtue or blinds so need around 150 steps to open the blinds. I also got it working with Alexa (for friends) but this was a pain and took a lot of trial and error due to poor instructions (see my closed issue). I just thought it would be a lot easier and better to use with the smart things hub but it wasn’t the out of the box experience I was expecting like it was with Alexa. I wish google would be more open so we can integrate directly with the esp chips. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

thehookup, 2019-09-21:
The ST integration works, but needs to be coded in arduino.  The WiFiManager version of it won't work due to the way that the STAnywhere library initializes the server.  

The .ino file for smart things is in this repo.  Unless I rewrite ST anywhere (not likely) or it is updated to work with wifimanager it will be the only way to use ST.

xHirscHx, 2020-02-05:
Same boat as above with ST bin. So i got it working with Alexa bin. but really want it to run in ST. 
If the wifi manager intructions dont work for ST. Are there instructions how to "but needs to be coded in arduino" Im more of a hardware guy and copy and pastier.
thx

CurtisFeatures, 2020-02-05:
I could never get it going with ST even using Arduino IDE. In the end iv gone back to MQTT and its all setup in my openhab server - I just like things to have hubs in case the openhab server goes down things still work.

ryancasler, 2020-06-25:
If it's only a question of a Child DTH for ST_Anything, I just need a clear understanding of what information the board is expecting to receive and what it is going to send back to the hub.  Is sketch expecting just a raw integer with nothing else before or after to identify it?  Or is it also expecting to receive the current position every time it receives a command?  What type of acknowledgement that the order has been received will be received back after the POST request is sent to the board?  Typically, that is when the event is created in ST to update the device in the ST database, not when the command is sent.  For example, when commanding a switch to turn on with ST_Anything, the HTTP POST will be sent to an endpoint like this:
`http://ip.of.board:8090/switch1%20on?`
The "?" is used as a built-in "end of message" character and if you don't include that, a stoock ST anything board won't recognize the command.  Conversely, if you are trying to use a stock ST_Anything DTH and the sketch is not expecting the ? at the end, the board won't recognize the command being sent to it.  A few more details on what http strings will be send back and forth is all I need to be able to code a DTH in Groovy for ST or Hubitat.  I can't completely tell since MQTT is still fairly new to me.  However, I am fairly familiar with ST_Anything (Hubduino) and have written several DTH for it.  Also, there are some example sketches in the ST_Anything library which only use the SmartThings ESP8266 Wifi library.  These might be of some help in adapting the sketch to more closely follow the standards.  I'll take a look at it later today or tomorrow and see what I can come up with.
I've just ordered a set of motors and drivers as I'm looking to change a couple of my servo controlled blinds over to running with the steppers in hopes of getting them to close more tightly.