thehookup/Motorized_MQTT_Blinds #6: New 3D Printed Servo Holder

cheechie, 2019-05-28:
My blinds (from Home Depot) are slightly different so I had to make a new servo holder.  Here is the STL file.  https://www.thingiverse.com/thing:3658592

tonydahtiger, 2019-05-28:
I am doing the same for my Bali wooden blinds.

cmille34, 2019-08-04:
Thanks @cheechie for taking the time to post your holder.  Worked great for my blinds as well.

tung256, 2019-10-20:
@cheechie , could you please take a picture of your blinds? i got mine from HD and i just want to make sure. also, when you print the STL file, are the units of measurement in mm, cm, or inches? thanks!

tung256, 2019-10-30:
i missed it by 1cm using the STL file he posted.  is that the same issue you experienced?
https://i.imgur.com/wSi4AtZ.jpg

cheechie, 2019-11-01:
can you just slide the servo closer.  Thats all I had to do.

Here is a picture of mine.  https://ibb.co/QrtFVKD

tung256, 2019-11-01:
it is already close but wont fit because the motor is 1cm higher than the hex stick.

tung256, 2019-11-03:
i just printed your modified STL file and i am happy to say it fits like a glove! wow.
here it is in its new home  https://i.imgur.com/D0F5GWl.jpg
thanks for posting.

v-yarotsky, 2020-04-26:
Thank you for posting @cheechie! And thank you @thehookup for this awesome guide!
I needed to make some tweaks for my home depot blinds, but so far so good :)
In case anyone needs them, here are my customizations: https://www.thingiverse.com/thing:4314465

echojjj, 2020-05-13:
Thank you @cheechie!  This fits perfectly!  Did you have to change the rotations from 12 to something else?

JoshuaGarrison27, 2021-01-12:
@cheechie Thank you. This fit snuggly in my blinds!

Davidadragoo, 2022-01-24:
This also solved my fit problem, thanks for sharing!