thehookup/Motorized_MQTT_Blinds #8: One NodeMCU for 3 blinds

TurboTronix, 2019-08-16:
Is it possible to control 3 blinds with on NodeMCU using other pins for the other 2 blinds? I have a bay-window with 3 blinds so was wondering if that was possible.

TurboTronix, 2019-08-16:
Also can this work with zebra still blinds?

cmille34, 2019-08-17:
You should be able to.  I have 2 blinds being controlled off the same pin and it works fine. If you wanted to control (3) separate stepper motors in (3) blinds you would need (9) PINs on the NodeMCU.  There should be enough pins to do that.  Just my 2 cents though - I'm not the author, just a big fan that really enjoyed following the instructions and automating 2 of our living room blinds.

thehookup, 2019-09-21:
It's easy to do if you don't want them to be individually controlled, just wire the 3 stepper drivers in parallel.  If you want separate control you'll need to fork the code and do a bit of modification.

closed 2019-09-21 by thehookup