thehookup/Motorized_MQTT_Blinds #13: Blind state is not retained and defaults to “unknown” On HA reboot/upgrade

vortex-uk, 2019-09-21:
Blind state is not retained by HA and defaults to “unknown” On HA reboot/upgrade. 
Only way to resolve is to send a state (that doesn’t match the blind’s current state) then HA realises the current blind position.

JoshuaGarrison27, 2019-10-25:
Is this why my motor only seems to turn in one direction?

thehookup, 2019-11-13:
I suppose I could change the state publish to retain: true which would allow HA to see it on restart, let me test it out.

Neo201069, 2020-02-05:
Still having this issue, any luck with fix?

mrwaltern, 2020-05-17:
In Home Assistant, I set the availability topic to be the /checkIn, with the payload_available to "OK," which is sent on your network every 90 seconds I also added a line to the .ino file in void checkIn to publish to the positionState topic every time it check in. It seems to solve the unknown issue for me. The only time this wont work is when the node MCU reboots, because its initial positionState playload is empty until it is run once.

sajovicd, 2021-01-28:
Hello, I also have problem with unknown position, is that problem realy so rare, or most of people just dont care about.
I would love to find solution for this. Is there some fix for that?
retain: true does not help.

sajovicd, 2021-02-11:
> In Home Assistant, I set the availability topic to be the /checkIn, with the payload_available to "OK," which is sent on your network every 90 seconds. I also added a line to the .ino file in void checkIn to publish to the positionState topic every time it checks in. It seems to solve the unknown issue for me. The only time this won't work is when the node MCU reboots, because its initial positionState playload is empty until it is run once.

Can youshow your code? PLEASE