thehookup/Motorized_MQTT_Blinds #15: One Direction Motor

JoshuaGarrison27, 2019-10-25:
Hey! I got everything wired up and it responds to the MQTT calls and everything but for some reason, the motor only turns in one direction. I have no idea what would cause it to do this. I made sure that I severed the middle contact point on the motor like in the video, double-checked my wiring, and made sure I didn't change much in the Arduino sketch. 

Any suggestions?

tung256, 2019-10-27:
i cant even compile the sketch and got bunch of errors as seen [here](https://drive.google.com/open?id=1-4yFY0elkVA2TDw5ZViSLAGbrf280FqB).   this is my first time with Arduino. any idea?
my additional board manager in preferences is: http://arduino.esp8266.com/stable/package_esp8266com_index.json.  
In board manager, i have esp8266 by esp8266 community version 2.5.2.

JoshuaGarrison27, 2019-10-27:
@tung256 make sure you have the right libraries installed and board selected

tung256, 2019-10-28:
yes. i got all those .h and .cpp files.
BUT i still cant compile and getting this error:  
C:\Program Files (x86)\Arduino\libraries\blinds\ArduinoOTA.cpp:189:34: error: 'U_FS' was not declared in this scope
     if (cmd != U_FLASH && cmd != U_FS)

did you have to edit this file **ArduinoOTA.cpp**?
thanks for helping me catch up and i will see if i have the same problems as yours.

tung256, 2019-11-03:
i used his "MQTT Only with WiFi Manager" bin file. no need to compile my own bin. works great! was able to go both direction. hope it works out for you

JoshuaGarrison27, 2019-11-05:
@tung256 I will give it a shot and see when I get a chance this week. That might be my problem.

Edit: Where did you find the precompile bin file? I only see the .ino files.

tung256, 2019-11-05:
the creator posted the bin file here : https://github.com/thehookup/Motorized_MQTT_Blinds/releases/tag/MQTT_WIFI
i followed the instruction and it works like a charm.  the bin file is on the bottom of that link.

JoshuaGarrison27, 2019-11-11:
@tung256 This still is causing mine to spin in one direction only for some reason.

tung256, 2019-11-12:
i bought all my stuff from the links he provided in the youtube vid.  the parts come in packs of 5. so i guess your next step is to try another component. try another motor, another controller, or another nodeMCU. mix and match, trial and error. 
here's 1 of my blinds working https://www.youtube.com/watch?v=R5MM3VOXV_I
i will ramp it up to have 1 nodeMCU controlling 2 blinds in the living room. fingers crossed they work

JoshuaGarrison27, 2019-11-12:
I agree. I have several of each component. I'm going to try a few more.

thehookup, 2019-11-13:
I'm guessing your DIR pin is not getting a good connection, double check it.

JoshuaGarrison27, 2019-11-27:
I have not forgotten about this. I will try it as soon as I can. I have been working on Christmas LEDs from your videos @thehookup

Justind-hub, 2019-12-24:
@JoshuaGarrison27  I had the same issue, after a few very frustrating hours I accidentally disconnected the wire on D5 of the nodemcu. Works great now!

JoshuaGarrison27, 2019-12-24:
I just wrapped up my Christmas light project. I'm gonna try this again this week or next. I'll try your suggestion @Justind-hub thanks

Justind-hub, 2019-12-24:
@JoshuaGarrison27 Don't bother it stopped working a few minutes later. Also that's the "sleep" pin, at 12v the motor gets pretty hot pretty quickly without sleeping.

thehookup, 2019-12-24:
I would not recommend disconnecting the sleep pin.  Your motors will burn out with constant voltage supplied to them.  It's possible that the driver you are using sleeps on pin low instead of pin high.  Change the line

#define DRIVER_INVERTED_SLEEP     1  
to 
#define DRIVER_INVERTED_SLEEP     0

and report back.  Are you using the drivers that I specified in the readme? https://amzn.to/2OZqW1W

Justind-hub, 2019-12-24:
@thehookup Hey Rob!

I didn't realize it was the sleep pin when it came unplugged by accident and started working magically, when i picked up the motor a minute or two later I realized what I had done by how hot it was. Lol.

I used all of the exact parts from the links, and have tried swapping out each one to make sure it isn't a bad motor/driver. 

The motor is sleeping correctly, I can spin it with my hands when it isn't "running", however I did try changing the line you mentioned with no success. 
As of now it jerks back and fourth for a few seconds when I open or close the cover in HA. For a while it was only spinning in one direction (which would flip on each restart of the nodeMCU).

Justind-hub, 2019-12-24:
All I can think of is that the stepper arduino library has been updated with breaking changes since you originally wrote the code. I can understand what most of it is doing but definitely don't know enough arduino code to be able to try it with a different library

Justind-hub, 2019-12-28:
I got it to work! After looking at the wiring diagrams for the stepper motor and the pin out for the driver I started crossing wires between the motor and the driver. The attached modified schematic works great for me!
[image]

JoshuaGarrison27, 2020-02-05:
So we disassembled everything and then rebuilt and reprogrammed everything. Magically works again. So not sure what we did different this time around.

closed 2020-02-14 by JoshuaGarrison27