thehookup/Motorized_MQTT_Blinds #16: Arduino IDE refuses to compile "Blinds_CONFIGURE.ino"

tung256, 2019-10-27:
this is my very first experience working with ESP8266 (i purchased using your affiliate link)

on Windows7, i use Arduino 1.8.10. In board manager, i have esp8266 by esp8266 community version 2.5.2. in Tools menu, i have:Tools -> Board -> NodeMCU 1.0 (ESP-12E Module)
Tools -> Flash Size -> 4M (3M SPIFFS)
Tools -> CPU Frequency -> 80 Mhz
Tools -> Upload Speed -> 921600
Tools-->Port--> com4

when compiling, i get this [error](https://drive.google.com/file/d/1-4yFY0elkVA2TDw5ZViSLAGbrf280FqB/view)

any idea what i did or did not do?

FILE> EXAMPLES> ESP8266> BLINK sketch works fine. i see my board's blue light on for 1 sec, off for 2 secs.
FILE> EXAMPLES> ESP8266WebServer>WebUpdate works fine too. i was able to visit its IP address

tung256, 2019-10-27:
i added more .cpp files   and not just the  .h files.
error messages decreased substantially from 160 to just 22. errors as seen [here](https://drive.google.com/open?id=1X1HGO1Qse-3qgOaGliUnJ2KhM2oygjAD)
what's next?

tung256, 2019-10-27:
C:\Program Files (x86)\Arduino\libraries\windows\ArduinoOTA.cpp:189:34: error: 'U_FS' was not declared in this scope

     if (cmd != U_FLASH && cmd != U_FS)

thehookup, 2019-11-13:
Looks like a bunch of libraries need to be installed, I'd recommend using the pre-compiled .bin file.

closed 2019-11-13 by thehookup