thehookup/Motorized_MQTT_Blinds #19: how to close 100%?

tung256, 2019-12-25:
i was able to close the blinds 100% manually, by twisting the plastic rod.
with automation in place, i see the blinds can only be closed about 85%. the motor really struggles and just cant. is there any way to get that last 15%? better motor? or lubricate the blind's internal metal mechanism?

towerclimber87, 2020-02-10:
Same request.

JoshuaGarrison27, 2020-02-14:
I suppose that it depends on the blinds but mine had a similar problem. It wasnt that the motor didnt have enough torque to make the rotation but rather I did not put in enough steps in the software. So it really comes down figuring that out and taking action from there by either a better motor or if there is a way to mechanically reduce the load on the motor.