thehookup/Motorized_MQTT_Blinds #20: Physical buttons

sajovicd, 2019-12-29:
Hello,

I found this very useful, but in some rooms like toalet I would like to have also buttons for close and open shades.
Is taht possible?
How do you solve the problem where you dont have your phone or tablet at the hand and want to close the shades?

JoshuaGarrison27, 2020-02-14:
It should be possible to wire up a momentary button that could do this. This is actually a really good idea since for most people you have to remove the twist rod mechanism.

virtualJonesie, 2020-05-22:
I'm working on this currently. I try to always have a non-phone, non-voice way to operate my smart devices. When I get this working, I'll post it here.

JoshuaGarrison27, 2020-05-22:
@virtualJonesie If you need someone to test, let me know. I just bought some momentary switches for this same issue.

virtualJonesie, 2020-05-22:
I bought some of these a while back. My goal is to make one sire open, and one side close. They're momentary, SPDT rocker switches, and they just so happen to fit in the factory hole for the tilt rod mechanism. No permanent modification of the blinds needed.

https://www.digikey.com/short/z4v1pv

My blinds are the 2" faux wood blinds form Home Depot.

https://www.homedepot.com/p/Home-Decorators-Collection-White-Cordless-Faux-Wood-Blind-with-2-in-Slats-10793478184453/206301010

virtualJonesie, 2020-05-23:
Here is my recreation of the original, with a SPDT rocker switch (linked above) included. I have not updated the code yet, but this should take care of the hardware side of things. Again, my plan is to use one side of the button for "open" and one side for closed, similar to the Home Assistant implementation created by The Hook Up.

[image]

virtualJonesie, 2020-05-24:
I made some very minor changes to the code. On button press, I have it publish either OPEN or CLOSE to the blindsCommand topic, depending on which button is pressed. Again, I'm using a SPDT momentary rocker switch, so one side is Open, one side is Closed. This way, the exisitng code handles the open/closing, and I don't have to check to see if the blinds are already in their desired state.

In the definitions section, I added definitions for the Button Pins
```
#define BUTTON_OPEN_PIN           D1
#define BUTTON_CLOSE_PIN          D2
```

In the setup() function, I set the modes to Input for those pins:
```
pinMode(BUTTON_OPEN_PIN, INPUT);
pinMode(BUTTON_CLOSE_PIN, INPUT);
```

And in the loop() function I added logic to react to button presses:
```
if (digitalRead(BUTTON_OPEN_PIN) == HIGH) {
    // Send a message to open the MQTT topic
    client.publish(USER_MQTT_CLIENT_NAME"/blindsCommand","OPEN");
}

if (digitalRead(BUTTON_CLOSE_PIN) == HIGH) {
    // Send a message to open the MQTT topic
    client.publish(USER_MQTT_CLIENT_NAME"/blindsCommand","CLOSE");
}
```

poggerb, 2020-07-04:
I wish I had found this earlier.....

I didn't have the coding knowledge, so improvised, by setting up some Sonoff Basics as the triggers for each of the blinds. Automation in HA to monitor the State and then Action Call Service to open/close covers.

I'll have to get some switches and try your modifications. Would have saved me some time money.... haha.

JoshuaGarrison27, 2021-01-12:
@poggerb You bought Sonoff Basics for each blinds installation? Wow. You were determined my friend. haha.

OOIOIOO, 2021-11-27:
Do not pull down low, most pins on NodeMCU are already pulled high. as such u set them as input, and that pulls them high by itself, so u send ground to your button and connect button output to that pin. in your code you check if the pin is Low ..... and .... bob is your uncle. there are no resistors involved, no power being sent anywhere and no sending 5v to nodemcu which runs on 3.3v.

Dannzig, 2022-01-17:
@virtualJonesie 
I too love automation, but have a wife and 2 daughters that get frustrated and need manual controls. Did your solution work? Do you have any video of the buttons in action?

rixmix123, 2022-01-18:
If you have zwave set up you could program a button from aqara for example to control them.

Peterpc58, 2022-11-26:
>I also upgraded the software a lot with a web server and such, if i have extra time ill post it one day, but that's going to be in future. 
When do you have the time to post your upgraded software?

JoshuaGarrison27, 2022-11-26:
> >I also upgraded the software a lot with a web server and such, if i have extra time ill post it one day, but that's going to be in future. 
> 
> When do you have the time to post your upgraded software?

My fork doesn't have physical button support but it has OTA built in. No Alexa support either. I just wanted to be able to update without taking down the blinds. So if that sounds okay to you here it is.... https://github.com/JoshuaGarrison27/Motorized_MQTT_Blinds