thehookup/Motorized_MQTT_Blinds #21: Microstep pins configuration..

manojjenago, 2020-01-05:
Hi,

I can see the blow pins are used in code for microstep , but I guess there is no wiring to nodemcu.. Please let me know how we can define pins for microstepping or it is only a number ?? Sorry if it is basic. I am pretty new 

#define STEPPER_MICROSTEP_1_PIN   14
#define STEPPER_MICROSTEP_2_PIN   12
Thanks,
Mj

echojjj, 2020-05-13:
My motor also goes forward and backward.

echojjj, 2020-05-16:
@manojjenago It's the motor wiring.  You have to find the two pairs.  One pair will be connected to A1 and A2 and the other pair will be connected to B1 and B2.  The wiring schematic included isn't correct for every motor.  This [https://user-images.githubusercontent.com/58923199/71539161-3fbe0780-28ed-11ea-80e9-8de35e4b15f4.jpg](url) @Justind-hub worked for me.

echojjj, 2020-05-16:
@manojjenago It's the motor wiring. You have to find the two pairs. One pair will be connected to A1 and A2 and the other pair will be connected to B1 and B2. The wiring schematic included isn't correct for every motor. This https://github.com/thehookup/Motorized_MQTT_Blinds/issues/15#issuecomment-569386423 @Justind-hub worked for me.

manojjenago, 2020-05-28:
After I changed to new motor and driver, it is working fine for me.
Thanks,
Manoj

closed 2020-07-08 by thehookup