thehookup/Motorized_MQTT_Blinds #22: Gearbox

mssmison, 2020-02-01:
You mentioned you designed some gearboxes, do you have those stl's around by chance still for the 6mm rod? These steppers work fine on my smaller blinds, but the larger blinds require too much force to tilt for these to handle.

mssmison, 2020-02-02:
Ignore the above. I printed this  ( https://www.thingiverse.com/thing:3642542 )4:1 gearbox (I only needed I layer) along with a custom planet carrier which has the 6mm rod attachment and it worked great.

closed 2020-02-02 by mssmison

dwalto, 2020-02-25:
> Ignore the above. I printed this ( https://www.thingiverse.com/thing:3642542 )4:1 gearbox (I only needed I layer) along with a custom planet carrier which has the 6mm rod attachment and it worked great.

Do you have the stl for the custom rod attachment?  I have two larger blinds where the stepper alone can turn the rod.

mssmison, 2020-02-25:
> Do you have the stl for the custom rod attachment? I have two larger blinds where the stepper alone can turn the rod.

Yeah for sure.

This was for my size rod which I think was 5mm. I found I still had to add a small bit of tape around the rod to get this to fit nice and tight. This is just the carrier, print the rest from the link above.
Also don't forget to increase your revolutions by 4x because of the 4:1 ratio.

[Planet_Carrier.zip](https://github.com/thehookup/Motorized_MQTT_Blinds/files/4248095/Planet_Carrier.zip)