thehookup/Motorized_MQTT_Blinds #27: Safer motor

Alex1602, 2020-03-04:
Hi,

In the video you mentioned that the motor can get hot because of the overpowered driver. Did you find a good alternative that should work without tampering the motor? I think the fire insurance would have an issue with the current setup.

So question is, are there better motors out there that do not have this issue?

sukhvinder1, 2020-05-11:
@Alex1602 Hey Alex, were you able to solve this issue?

I was working on this project, everything works fine just the motor is getting super hot in 2-3 mins. 

Any suggestions for me ?

Alex1602, 2020-05-12:
Unfortunately not, I did not start on this project until I get a better motor suggestion :)

sukhvinder1, 2020-05-12:
@Alex1602 Thanks for replying

@thehookup Can you please give suggestion to us ?

christensonautohome, 2020-05-31:
My blinds run for a max of 10 seconds. If you motor is running for 2 or 3 minutes then it’s wired wrong. If not in use then there is no power draw and the motor is room temperature.

thehookup, 2020-07-08:
There is no fire issue with this motor if you have it installed correctly.  I've run this motor for over a minute continuously at 12V and it doesn't get too hot to touch.  The only problem would be if you haven't wired the enable pin correctly.

closed 2020-07-08 by thehookup