thehookup/Motorized_MQTT_Blinds #28: Questions from a n00b - First Arduino/NodeMCU project

k0bayash1maru, 2020-03-05:
This is my first ever DIY with a Arduino/NodeMCU. 

I have a few questions on the project, I am a complete n00b so please excuse any lame questions :) 

**Prereqs**:
```
1. I've bought all the parts from the links you posted with crimper set and a soldering kit 
2. Watched your videos on Arduino and Home Assistant/Node-RED  and of-course the one on the blinds multiple times 
3. Installing Arduino and Home Assistant and Mosquito broker on my always on Ubuntu server 
4. Have the 3D parts printed
5. Have a few CAT 5 cables lying around
6. Have an Alexa device, Alexa app on an iPhone
```

**Question**: 
1. I looked at the wiring diagram you added but this is still very confusing to me. I plan on doing this project for the 3 blinds in my bedroom like your setup with 3 motors going to a single NodeMCU: 
```
1a. You said you connected the wires from each blinds together before attaching them to the NodeMCU - how did you do that ? 
1b. What does the wiring for one motor + one NoedMCU look like ?
1c. What is the wiring when you attach them together ?
1d. How is power hooked up to these ?
1e. What are the clear cases with the orange tabs that you connect these ? (where can I buy them ?)
1f. You said "need to make sure that the connector is in the right orientation" - What does this mean ? 
```
Can you please send some pictures of the above wiring I think that will help me a lot! 

2. I plan on using this with Alexa:
```
2a. Do I just upload the  Blinds_CONFIGURE_Portal_Alexa.ino via Arduino to NodeMCU (after making the config changes) or do I need to do anything with the .bin file ( I assume its an either or)
2b. Do I need to upload Blinds_CONFIGURE.ino ? I assume I use this  along with Blinds_CONFIGURE_Portal_Alexa.ino and this will control the blinds via home assistant.
2c. I assume I don't need Node-RED for this - is that right ?
```
`3. Upload the HomeAssistant config with the YAML file`
`4. Anything else that I need to look for from an absolute beginner's prospective ?`

Any help greatly appreciated!!!

Thanks in advance

ImpossibleCodes, 2021-04-04:
I'm wondering the same thing, I wired it up using the schematic on a breadboard and ended up emitting magic smoke, If you or someone else has figured it out and has pictures or something to help, that would be amazing

k0bayash1maru, 2021-04-04:
Try this schismatic:
https://github.com/thehookup/Motorized_MQTT_Blinds/issues/15#issuecomment-569386423