thehookup/Motorized_MQTT_Blinds #36: SCAD File for BlindsBase

dsandor, 2020-06-20:
It would be very helpful to have access to the OpenSCAD file for the `BlindsBase.stl` file. It would be useful for minor variations in blinds. For example by blinds need about 8mm trimmed off the top of the assembly.

p-doyle, 2020-06-29:
Yes, please!  Current design doesn't work with my blinds either and I was hoping not to have to design it from scratch.

dsandor, 2020-06-30:
I created one on my own. Mounted it in 2 blinds and it seems to work great. I had to put a tiny bolt through the metal to act as a backstop. The motor would work its way back and detach after about 20 cycles. The bolt was a super simple way to prevent that. I suppose you could make it wider to be a tight fit but this worked well for me. 

https://gist.github.com/dsandor/c1f7e3ffebb0b491225a713f0423aede

Enjoy.

thehookup, 2020-07-08:
Thanks dsandor, added your SCAD file to the file list.

closed 2020-07-08 by thehookup