thehookup/Motorized_MQTT_Blinds #37: Fault Pin Connected to 5v?

ryancasler, 2020-06-25:
I am wondering why you have the DRV8825's Fault pin connected to 5v.  From what I can see from other schematics that use this driver, that pin is never connected to the VCC of the microcontroller.  The datasheet says that it is output low when there is a fault detected.  Why would you want to short out the power supply if there is a fault detected?  Won't that burn out your supply?

echojjj, 2020-07-01:
I was wondering the same thing.  I think the DRV8825 GND is to common to the microcontroller ground, and that be the mistake in the wiring diagram, but it does work even with 5v to fault pin.

ryancasler, 2020-07-03:
And doesn't without the 5v to the fault pin.  I am wondering if it has something to do with using the Enable pin rather than the sleep pin to power down the motor controller when not moving.  I decided to just try it since I have 5 of each and only 3 windows where I'd possibly use them.  So, even if I blew one, I'd still have one as a backup.  :)  Glad I wasn't the only one that was confused by that.

thehookup, 2020-07-08:
That is really strange.  The reason I wired it that way is that I'm used to working with A4988 drivers that take Vdd on that pin.  I've always heard/read that the DRV8825 can be wired interchangeably with the A4988, so that's what I did.  It hasn't caused any issues for me.  It looks like the only other pin to use to supply Vdd is the RST pin?  That's strange to me.

closed 2020-07-08 by thehookup

ryancasler, 2020-07-08:
Yeah, it is definitely required to have the pin connected to 5v to get it to work.  I just can't figure out why either.  I it must have something to do with using the EN pin instead of the SLEEP pin to shut the board down.  

When it comes to the tiny current adjustment on the board, when you said in the 12 o'clock position, what did you mean?  when oriented in which way?  Did you mean half way between min and max?  I have only 3 blinds to automate and 5 motors but I'd rather save them for other screwups later on than to blow one up unnecessarily.

echojjj, 2020-07-08:
[image]
[https://www.pololu.com/product/2132](url)

echojjj, 2020-07-08:
@ryancasler the position of the pot would be at 12:00 when the flat edge is at 6:00.  look closely for the flat edge.

echojjj, 2020-07-08:
@ryancasler the pot is at 12:00 when the flat edge is at 6:00.  look closely for the flat edge.  here's a picture of a pot at 12:00
[image]

softtechs1, 2022-07-18:
I got everything wired per the diagram, programed MCU and getting data from MQTT. Just can't get the motor the move at all. Does the GND on the DRV8825 need to be connected to the GND of the MCU? From looking at the wiring on the A4988 driver, its connected to the microcontroller.

ryancasler, 2022-07-20:
How are you powering the board?  Are you using a buck converter like in the diagram or are you using a 12v supply and a 5v supply.  I have mine wired like the pic and it works.  You just have to make sure the two grounds are tied together somewhere.  No ground, no current flow.

softtechs1, 2022-07-20:
Thanks so much for responding! I purchased everything from the amazon links and wired as in the pic schematic. Using the 12v power supply, connected to the input side of the buck converter. Ran another 12v wire pair to connect the DRV8825’s UMOT/GRN pins. NodeMCU 12E gets power from the buck’s 6V output side. From the buck converter output side, I also have another 6V pair, which only the + from the buck converter is connected to the DRV8825 FLT pin, the negative is not connected as per the schematic, which is confusing. Everything else the schematic should be straight forward (Step, Dir, EN and 4 Motor wires). From what’s in the schematic, I don’t understand how the grounds tie together?
See my test bench photos.

[image]
[image]

ryancasler, 2022-08-01:
First of all, the output of the buck converter should be set to 5v.  Not 6.  
Second, how do you have the wires coming out of the Buck converter.  one side is + and the other is -.  So you need one from each side to the board.  You have two of the wires for the motor backwards.   The orange and pink should be pink-orange.
What type of 12v power supply are you using?  How many amps does it put out? Have you adjusted the current limiter on the DRV?  If that is all the way down, your motors won't move.
If you disconnect the buck converter's + line and just keep the gnd attached, you can plug in your board to your PC and see the output of the Serial log so you can try and see what is happening.  That way you can tell if it is able to connect to your wifi and actually receive the MQTT message.

softtechs1, 2022-08-01:
@ryancasler I got this all working by replacing a bad DRV8825. Seems I received a bad batch along with motors. Installed 2 blinds so far and working on the 3 more. Thanks!