thehookup/Motorized_MQTT_Blinds #38: Cannot Compile ST Sketch

ryancasler, 2020-06-25:
The Sketch Uploaded for ST will not compile as written.  Right off the bat it gives an error that the IP is formatted incorrectly at the top of the sketch.  There are too many decimal places because it is treating it like a number.  How did you ever get this to compile?

ryancasler, 2020-06-26:
After many modifications I was able to get the sketch to compile and connect to ST.  I get all my motor components on Monday.  If everything works, I'll send a PR to update to a working version.  I should also have a DTH/ Hubitat driver as well.  I have one working in dry run with no motor attached.  So, if wiring is as easy as you say, should just be able to plug it in.  Fingers crossed!!

closed 2020-06-26 by ryancasler

Fran6u, 2021-06-16:
Hi. I am new at coding and HA and facing similar issues, can you possibly post a sample of the modifications you made.

Fran6u, 2021-06-16:
> After many modifications I was able to get the sketch to compile and connect to ST. I get all my motor components on Monday. If everything works, I'll send a PR to update to a working version. I [...]

Hi. I am new at coding and HA and facing similar issues, can you possibly post a sample of the modifications you made.