thehookup/Motorized_MQTT_Blinds #40: Integer Position

ledrewski3, 2020-08-30:
I'm having trouble running the motor for the first time. It won't move. Not sure if I'm sending the topic the right way?

millerj1119, 2021-03-02:
I am having this same problem.  I have all of the connection correct and flashed my 8266 with the needed information.  Downloaded the MQTT Explorer and everything seems to be sending and receiving correctly. but no motor movement.  Any ideas?

kboghdady, 2021-11-29:
I am having the same issue

T00LF00L, 2022-06-15:
A bit late here, but I just built this project and had the same problem. I can tell you what’s wrong, the wiring diagram is incorrect. Wire the stepper drivers this way and it will work.
[image]

webxl, 2022-07-23:
This did the trick for me, thanks!

T00LF00L, 2022-07-23:
@webxl Awesome, glad I could help someone! I’ve since done a little more research and learned this driver was updated at one point and the pin out changed. The driver pictured in the wiring diagram is the new version, but the wiring as shown is in fact correct for the old driver version.

reinervl, 2022-08-17:
@T00LF00L this new wiring diagram fixed it for me as well., so thank you! However, I may have a new issue. Every time the nodeMCU loses power and restarts after power is back, the motors won't move until I unplug them and plug them back in. It's as if the nodeMCU can't handle the motors being connected at start-up. Any ideas on how to remediate this?

reinervl, 2022-08-26:
@T00LF00L thanks for your help. I should also mention that I have 4 stepper motors and drivers (1 driver per motor) connected to one nodeMCU. When I leave them all connected and I plug in power, all 3 pins (EN, STEP, DIR) read 1.54V so floating? When I disconnect all motors/drivers and then plug in power everything works and STEP and DIR read low (0V) and EN reads HIGH (3.3V).

T00LF00L, 2022-09-02:
@reinervl To be perfectly honest with you.. this is the first circuit I've built in 30 years, the last time I wrote code was on a Commodore 64 and this project took me 4 days of troubleshooting to finally get working with Home Assistant and HomeKit.  Coincidentally I just had one blind do the exact same thing you described, but it's only happened that one time.  I did some Google.. it says these three pins have a 'weak' high during boot, but they can be high or low and it won't affect boot.  So the only other thing I could find is maybe it's a power delivery issue during boot?  Sorry I can't really be much more help as electronics were very different in the 80s, you may want to open a new discussion? I'm still trying to figure out why mine occasionally open back back up at night..