thehookup/Motorized_MQTT_Blinds #41: Blind keep closing by there-self

jhppc, 2020-09-02:
this is driving me nuts. I dont have any automation set up to close the blinds but the blind keep closing my there self.

[image]

rjmcfadd, 2020-12-07:
This is happening to me as well! Any advice @thehookup?

JoshuaGarrison27, 2021-01-12:
I'm having this issue too. I think it is because of how the code reacts when it experiences a fault, reconnects to wifi, or MQTT checks in (such as with home-assistant or node-red). I am going to try to look into this myself since Rob is pretty busy.

If I am successful finding a solution, I will try to open a PR for Rob or I will at least put some notes in this issue.

rjmcfadd, 2021-01-12:
I actually just fixed my issue. The problem I was having was the blinds would open if I lost power. I used this rule:

Rule2 ON Power1#Boot DO TuyaSend4 1,1 ENDON
Rule2 1

JoshuaGarrison27, 2021-01-12:
Are these Tasmota rules? Or something else?

rjmcfadd, 2021-01-12:
Yeah, it is Tasmota rules.

JoshuaGarrison27, 2021-01-12:
You got Tasmota to work with this setup? Do you mind sharing your configuration string if so?

ericullmann, 2021-01-19:
I am having the same issue as well. Did anyone find a solution to this?

JoshuaGarrison27, 2021-01-19:
@ericullmann I am rewriting some of this code because I couldn't get it to work with Tasmota without the motor getting stupid hot. So once I get my code cleaned up and tested to not auto open I'll will share it.

ericullmann, 2021-01-20:
Appreciate it!

JoshuaGarrison27, 2021-01-20:
@ericullmann I have begun trying to track down this bug but I cant seem to reliable reproduce it. Do you have any specific steps that you do that make it happen? Here is my fork of this project: [https://github.com/JoshuaGarrison27/Motorized_MQTT_Blinds](https://github.com/JoshuaGarrison27/Motorized_MQTT_Blinds)

thehookup, 2021-01-21:
I'm super confused by this thread.  Is this an issue with Tasmota, or my code?  Is this issue with the MQTT version or the Alexa version?

ericullmann, 2021-01-21:
My issue is related to your code and seems to be affected after version 0.118 of home assistant but does seem to be related to reconnects of the nodemcu to broker.

thehookup, 2021-01-21:
I'd suspect 2 issues: First, there are two ways to open the blinds with MQTT via HA 1) send the OPEN command and 2) send a position value.  If HA changed the way it publishes positional commands it's possible that there is a retained OPEN command or position command that gets picked up on reconnect.

ericullmann, 2021-01-21:
@thehookup  I would need to rennable the position topic as I have disable every topic except the command so now position is missing on home assistant for the blinds. However, i can tell you the problem still occurs and is random. It affects certain blinds and then sometimes they all are fine.

thehookup, 2021-01-21:
Do you have a way to check for retained messages? (MQTT Explorer, Node Red, etc?)

I've got 9 of these deployed my my house and the most recent HA, no issues.

rjmcfadd, 2021-01-21:
> @rjmcfadd You got Tasmota to work with this setup? Do you mind sharing your configuration string if so?

Sorry, I wasn't getting notifications for this. I just realized that I posted in the wrong place. I'm using the Zemismart Blinds Controller Motor thehookup recommended in one of his videos that I flashed with Tasmota. When I searched my issue I found this.

ericullmann, 2021-02-02:
@JoshuaGarrison27  Its is completely random but It does appear to be on reconnects as well. At least this is what I can tell. It happens at random times and random blinds as well. The strange thing is that rolling the HA code back to 0.118 resolves this but HA says they have made not changes on MQTT.

JoshuaGarrison27, 2021-02-02:
@ericullmann Mind sharing your yaml for home assistant to your blinds? I have not had any issues recently but I am running my own modifed version of this code. So if you give me your configuration, I'll try it on my blinds to see if the bug still happens with my code.

ericullmann, 2021-02-02:
Sure do you want the blinds code and the home assistant yaml or just the ha yaml?

JoshuaGarrison27, 2021-02-02:
The HA Cover MQTT yaml you are using for one of your blinds.

ericullmann, 2021-02-02:
@JoshuaGarrison27  Here is it

  - platform: mqtt
    name: "Living Room Blinds"
    command_topic: "livingroom/blindsCommand"
    set_position_topic: "livingroom/positionCommand"
    position_topic: "livingroom/positionState"
    state_topic: "livingroom/positionState"
    retain: true
    payload_open: "OPEN"
    payload_close: "CLOSE"
    payload_stop: "STOP"
    position_open: 0
    position_closed: 12

  - platform: mqtt
    name: "Dinning Room Blinds"
    command_topic: "dine/blindsCommand"
    set_position_topic: "dine/positionCommand"
    position_topic: "dine/positionState"
    state_topic: "dine/positionState"
    retain: true
    payload_open: "OPEN"
    payload_close: "CLOSE"
    payload_stop: "STOP"
    position_open: 0
    position_closed: 12

JoshuaGarrison27, 2021-02-02:
@ericullmann Thank you! I have applied these settings to one of my blinds. I want to see if it makes any kind of difference. I was using a more dumbed down version of MQTT for my blinds. I didnt care to have position and such in HA. So my early guess would be that it has something to do with the positionCommand/positionState in conjunction with retain=true. I will share if I start having some issues. My plan it to also try to record all MQTT commands related to this device id. This will hopefully give some more insights to the issue. 😄

ericullmann, 2021-02-03:
I did do a test where I did retain false and only use the command topic but the result was the same.

thehookup, 2021-02-03:
Can you try this YAML and report back?

platform: mqtt
name: "Living Room Blinds"
command_topic: "livingroom/positionCommand"
set_position_topic: "livingroom/positionCommand"
position_topic: "livingroom/positionState"
state_topic: "livingroom/positionState"
retain: true
payload_open: "0"
payload_close: "12"
payload_stop: "INVALID"
position_open: 0
position_closed: 12

platform: mqtt
name: "Dinning Room Blinds"
command_topic: "dine/positionCommand"
set_position_topic: "dine/positionCommand"
position_topic: "dine/positionState"
state_topic: "dine/positionState"
retain: true
payload_open: "0"
payload_close: "12"
payload_stop: "INVALID"
position_open: 0
position_closed: 12

ericullmann, 2021-02-03:
@thehookup I have made the modification and I will report back.

ericullmann, 2021-02-03:
@thehookup  same result.

JoshuaGarrison27, 2021-02-03:
@ericullmann Is there anything in particular that you find makes it happen? Or was it coincidence that it happened within 20 minutes?

ericullmann, 2021-02-03:
It was coincidence. The only thing I did do at the same time is tell Google to turn off a room that is not associated., The one thing I can say is in home-assistant when these events happen if I look at the history say for the dine blinds I see it was listed as "was closed" at 828 vs. what I woudl see if I did it or google did it would be service..cover.set_cover position or if I do it by the app it says closed by eric. That the only thing that is consistent when this happens.

ericullmann, 2021-02-03:
I can say with the change I made this morning in particular my dinning blinds have closed over 10 times on their own today.

thehookup, 2021-02-03:
I highly suspect a retain issue here.  It would also fall in line with more problems after the change you just made.  My guess is you have "OPEN" retained on "livingroom/blindsCommand" and HA stopped sending that command after the update and now only sends on position command, or something along those lines.

ericullmann, 2021-02-03:
So how would I change this if this is the case as I didn’t modify your original code on the nodemcu so I’m thinking that where this modification would need to occur?

JoshuaGarrison27, 2021-02-03:
@ericullmann I have run your yaml all day with no issues. All my automations ran successfully as well.

MQTT Logs for your YAML you provided me yesterday: [https://pastebin.com/pzGnvfLs](https://pastebin.com/pzGnvfLs) 
As you can see, most of what is happening is the `checkIn` periodically. However, there tends to be a random `positionCommand` sent from time to time. No idea why just yet. I have not touched nodered, ha, or docker since last night after setting up your YAML and this logger. I wont pretend to understand how `retain` works within the broker and such.

**Be aware:** I am using a modified version of Rob's code. So if you want to see what I have you can check out my [repo](https://github.com/JoshuaGarrison27/Motorized_MQTT_Blinds). Only real difference so far is that I moved away from ArduinoOTA (because I could not get it to work) to using `ESP8266HTTPUpdateServer`. I also plan to add a landing page for the device so you can quickly see what the MQTT name is and such at a glance.

I know this isn't very helpful and I'm sorry. I just want to try to be as helpful as I can be.

Something you can try (if this doesnt break your workflow) is to simplify the YAML a bit and make them a little "dumber" with something like this: 
```
platform: mqtt
name: "Kitchen Window Blinds Left"
state_topic: "Blinds001/blindsCommand"
command_topic: "Blinds001/blindsCommand"
value_template: >
    {% if value == "CLOSE" %}
    closed
    {% else %}
    open
    {% endif %}
payload_open: "OPEN"
payload_close: "CLOSE"
retain: false
```
This way, HA is only monitoring for the `blindsCommand`. And it only gets that when you actually control them. So if you are getting random `positionCommand` messages, then HA doesnt know about it and wont act on it. Just an idea.

ericullmann, 2021-02-04:
Also I should note that it will randomly open sometimes when it is closed and when it is open it will close so I’m with you with retain being an issue but I’m not sure exactly what to fix.

ericullmann, 2021-02-04:
I just tried this abd it can open the blinds but when I try to close them it doesn’t close

ericullmann, 2021-02-04:
@JoshuaGarrison27 So I got your yaml to work by slightly modfying the open and close payload to 0 and 12 respectively. It appears to be working but I need to watch this further. However, the status in HA always shows "open" it doesnt update when I close to a closed state.

ericullmann, 2021-02-04:
@JoshuaGarrison27 @thehookup  I tried your code and they did close on their own as well. I think this is with the reconnect and the code itself on the nodemcu. I am thinking it is with the checkin but I cant seem to find exactly what is causing it.

JoshuaGarrison27, 2021-02-04:
This is so bazaar. Mine has been working flawlessly. Can you give us some more information on your setup? Such as for me, I run HA, NodeRed, and Mosquito MQTT all in separate docker containers.

ericullmann, 2021-02-04:
Sure I run mosquito mqtt on a raspberry pi. I run ha on another raspberry pi in a docker container. Then I have 10 nodemcu that are running my blinds and another 3 nodemcu running temperature sensors abd sending data to mosquito.  

Thanks
Eric

ericullmann, 2021-02-04:
I may have found the issue and will do some more testing. I will report back tomorrow with the results.

ericullmann, 2021-02-06:
@JoshuaGarrison27 @thehookup  Thank you both for your help. I appreciate it. I was able to solve the problem by upgrading my MQTT broker and everything has been stable for two days now so I think we are good.

JoshuaGarrison27, 2021-02-06:
@ericullmann Awesome. Glad it was something simple. My mqtt container got updated to 2.x yesterday and my whole mqtt network went down. So be aware if you are using the mosquitto MQTT container.

craigdthom, 2021-04-26:
> posted this earlier ... I was having a similar issue

 I have noticed different behaviors of the NodeMCUs -  https://imgur.com/8FNR8mx

    When power to the NodeMCU is lost .... some of the NodeMCUs reset( Change state - Open/Close) and others do not
    When my Home assistance server is restarted .... some of the NodeMCUs reset( Change state - Open/Close) and others do not
    When WiFi coverage changes or drop(Mesh WiFi was a headache) ... some of the NodeMCUs reset( Change state - Open/Close) and others do not

I 'm very stable now with just the two routers - 1 upstairs and one downstairs ( I am controlling 25 blinds without any issues )

-Craig

petereit, 2022-05-15:
@ericullmann  What broker/version are you using. I'm Mosquitto broker (6.1.2) and am still experiencing the issue.