thehookup/Motorized_MQTT_Blinds #45: Are 28BYJ-48 motors strong enough?

nuaimat, 2020-11-15:
Hello and thanks for the amazing detailed video, I truly appreciate the effort you put into this,

After buying the items and following exactly your instructions, I'm facing this problem, seems like the 28BYJ-48 motors are not strong enough for my blinds https://youtu.be/iennyIrTf10 truth is, I tried the same setup on another small window blinds like 1/3 the size of these in my video, and they have the same problem, any idea what might have gone wrong, or do you have a recommendation for a stronger motor? 

PS: one thing I was suspecting I might have done wrong was converting the 28BYJ-48 from unipolar to bipolar. but I build a small circuit with LED and get sure that the common wire (red) was disconnected so I am sure that was not the issue. 
 
Thanks

JoshuaGarrison27, 2021-01-12:
@nuaimat I cannot speak for your issue directly. But it works perfectly fine for my blinds. Are your blinds more "premium" with actual wood (or fake wood) slats? If so, it could very well be a weight/torque issue. However, it looks like in the video that you are getting some movement, so I wonder if it has something to do with some kind of slipping with however you mounted your 3d printed adapter. Maybe a video of the inside while it is trying to open would help diagnose it?

Here is what I can tell you with certainty...I have the 28BYJ-48 motors and they work great for my blinds.

nuaimat, 2021-01-12:
@JoshuaGarrison27 Hey Joshua, thanks for your input, it's totally a great idea, I think I should put a small camera in there and report back. 
I gave up on 28BYJ-48 right after this issue was opened, I tried using Nema17 with Tasmota but still having similar issues.

Whogives23, 2021-01-12:
@JoshuaGarrison27 . Ensure you have the polarity of both coils wired correctly. Otherwise they would counteract each other.

I have a similar(if not larger) being driven by two 28BYJ-48 motors.
https://photos.app.goo.gl/2onbQ6M7vUqVC6Wh8

xdenneboom, 2021-02-05:
Were you able to figure out the issue? I have the exact same thing happening, and was wondering if the motor was strong enough too. After a few milimeters of movements, I get the exact same sound without any movement, just like yours. 

My blinds are similar in width, but extend all the way to the floor. It does seem others have bigger or similarly sized blinds working effortless.

I have controlled the motor without hooking it up to the blinds, and seen the rotation happening without those sounds and issues. So it's definitely something that only happens when attached to the blinds.

OOIOIOO, 2021-11-27:
@Whogives23 

is there a reason you are using the "non converted" motor and the controller it comes with?  Also do u have an issue with overheating? I got an impression motor was heating up like crazy when on that controller

Dannzig, 2022-01-17:
> @JoshuaGarrison27 . Ensure you have the polarity of both coils wired correctly. Otherwise they would counteract each other.
> 
> I have a similar(if not larger) being driven by two 28BYJ-48 motors. https://photos.app.goo.gl/2onbQ6M7vUqVC6Wh8

What is that Manual button you are using yo open and close them?

JoshuaGarrison27, 2022-01-18:
> What is that Manual button you are using yo open and close them?

I actually abandoned the idea of having a manual button. I was running out of time on my vacation to install and everything. 😂

petereit, 2022-04-25:
On my first activation, I had the exact same problem. Two things helped:

1. While I had the blinds down, I put some lithium grease on every round plastic bushing that snaps the rod into place and worked it back and forth until rotating the rod became almost effortless.
2. I made sure the very bottom of the blinds was resting on a firm surface (my blinds are mounted inside the window casing, so the bottom rests on the window stool) so there was just a bit of slack on the lines so that the motor doesn't have to strain against the entire weight of the blinds.

dmasters74, 2023-07-28:
Hi 

Can anyone recommend a replacement motor with a more holding torque. My blinds fully close with this motor but the weight of the blinds pulls the motor back open causing the blinds to open slightly.

petereit, 2023-07-28:
Ideally, your blinds should be secured at the bottom so there’s no weight pulling on either side.