thehookup/Motorized_MQTT_Blinds #47: Connecting to MQTT broker, but MQTT broker doesn't see it

yetihahn, 2021-01-23:
I love this project. I have watched the video about 6 times. Read and re-read the directions. However, I still can't seem to get my MQTT broker to see the ESP. 

I flash the ESP (using tasmotizer) with the .bin file, and enter all the details, the ESP connects to the broker (as seen in the monitor via Arduino IDE). However, in HA I cannot see the ESP. I'm looking in configuration>instances>MQTT>"10 devices"
My other MQTT devices are there. But not the ESP. 🤔
My MQTT experience is limited Tasmota using MQTT with "setoption19 1" and it then appearing in HA right after. 

Troubleshoot I have tried:
Flashing the .bin via tasmotizer (making sure to erase content on the ESP)
Using the Arduino IDE to upload the sketch to the ESP
Trying 3 ESP8266s (two 12-E and a Wemos D1 mini)
Flashing the Alexa .bin to see if Alexa can find it. No luck. 
Using the other Alexa .bin. Still no luck. Left it connected over night to see if it "magically" gets found. (has happened with other projects.)
Leaving the ESP flashed with MQTT version powered overnight to see if it "magically" gets found by MQTT. 
Using Fing and my router's app to ensure the ESP is on the network and has an IP address. 
Putting in the wrong details for MQTT to see if the ESP still reports connecting to the broker. It doesn't claim to connect to the broker with the wrong IP, or password, or username. So I know it's not claiming to connect by default. 

I'm guessing there's a way to have my broker find the ESP. But, despite hours of googling, I can't figure it out. 😞 if anyone could point me to a "MQTT for dummies" (I've watched Rob's, but wasn't able to find what I needed) or point out the flaw in my methods I'd really appreciate it.

yetihahn, 2021-01-23:
Also I've made sure "enable newly added entities" is turned on.
I've toggled it off and on again to see if that "shocked" it into finding the ESP.

xHirscHx, 2021-01-25:
I only was  able to get the ALEXA  bin to work.  I was able to go threw alexa to smarthings then to hassio with a simulated blinds setup. I dropped that setup and now running tasmota  blinds and anther 2  blinds running on esphome threw hassio. Hours of trying to get a moter (use tasmota) or stepper moter( use esphome) to wotk.

xHirscHx, 2021-01-27:
Save your self of hours of Google time( trust me)  the hookup blinds work great with alexa but i needed local. look at this -----https://tasmota.github.io/docs/Blinds-and-Shutters/  same setup you already have and im guessing you already have tasmota devises.

yetihahn, 2021-01-27:
This looks good, I'll have to dig into this! I like the idea of using Tasmota as I'm already familiar with it. And yes, I've been Googling for hours...

I ended up figuring it out. I'll post what worked in case someone else needs this in the future. 

With the Alexa files I had to reboot the Echo and ESP8266 to get it to work. (Had some help.) I also waited over night between setting up and rebooting, which may have helped?

With the MQTT files I ended up downloading MQTT Explorer to understand what's happening with MQTT. This helped me see the states of devices and messages as they were sent (kinda.) It also let me publish and see how it changes things on the network. This took some trial and error, and caused me to learn a lot. 
When going into HA>Configuration>Integrations>MQTT># devices I still don't see my window blinds. I'm guessing this is due to the integration only showing MQTT devices that have states published to topics. The file Rob wrote doesn't do that, so I was confused why the window blinds weren't appearing in HA. Looking with MQTT Explorer and seeing the motor blinds listed there caused things to click for me. 
After adding the Cover (provided by Rob) in File Editor>configurations.ymwl I was able to publish positions to the device/topic by going to HA>Configuration>Integrations>MQTT>configure>publish a packet. For topic I can put: 
*nameOfMQTTDevice*/positionCommand  
For payload I put "1" then clicked "publish" then put "12" as the payload and clicked "publish" again. This got the motor to move. (Rob explained that their first value sent is the "home" position, so that part wasn't confusing, but mighy be over looked by others.) Learning to publish with MQTT was the tricky part for me. I practiced publishing using MQTT Explorer while seeing the states change and messages being sent, and was able to get the hang of how MQTT works. I'm very visual so not seeing states in HA really threw me off. 

Hopefully this helps other in the future!

closed 2021-01-27 by yetihahn