thehookup/Motorized_MQTT_Blinds #50: Motor Temperature

CalgaryCM, 2021-02-16:
Hey, Can anyone confirm if the motor gets HOT if the NodeMCU/ESP8266 is not connected?

I am starting this amazing build and during voltage checks I notice that the motor gets very hot when the nodeMCU/esp8266 is not connected. My initial gut feel (and prob incorrect as I am just starting) is the nodeMCU using the enable pin to sleep the motor. My concern here is if something were to happen to the nodeMCU it would stop sending "sleep" to the motor and the motor would heat up to the point of it being a fire hazard. 

I did verify my wire connections several times. I assume the wire color code on the diagram matches the wires on the motor and I have them connected properly. 

TIA!

Smitty357, 2021-03-18:
Mine does as well. I have tried multiple things with the same results.
I have had mine in use for several months working great, but decided to modify it because it seems to have connection issues at times. During this testing is when I found it gets extremely hot. I thought the enabled pin would have turned off when not used. Maybe the board is waiting for some feedback on the position to know it is complete?

OOIOIOO, 2021-11-27:
I'm still in the process of working on this project, but I had an episode where I killed node mcu and the controller by shorting probably and didn't realized that and put the blind up. After an hour or so I realized that the blind doesn't work very well so I took it down. It turns out it was soo hot that it melted the adaptor.

So my advise to anyone who is doing this is double sleep the motor through the Fault pin, by connecting it to one of the pins on nodemcu(instead of 5v directly), then setting the pin to High when needed and to Low otherwise, this way controller is completely off, and u don't have a risk burning your house. :)

rixmix123, 2021-11-27:
Can you provide a diagram of that double sleep set up? Everything for me works perfect, but I did disconnect the node mcu to test and the motor started to heat up. Thanks!

rixmix123, 2021-11-27:
> Can you provide a diagram of that double sleep set up? Everything for me works perfect, but I did disconnect the node mcu to test and the motor started to heat up. Thanks!

@thehookup any ideas for a good fail safe of the node does fail eventually?

OOIOIOO, 2021-11-27:
@rixmix123 
[image]
its fairly simple really, note the Orange line from D8. Its mostly software, the way I did it, I set it as output and during setup I set it to LOW, than any time I run the motor I set it to HIGH, and in the part where the moving is set to false, I set the pin back to LOW. So Controller is essentially always off and never heats up.
Now technically gpio pins 1 to 15 I think are all pulled HIGH so there is a chance that during boot the controller will on until the code kicks in, I believe gpio 16 is defaulted to LOW, so u could try that to be extra safe.

OOIOIOO, 2021-11-27:
> @rixmix123 [image] its fairly simple really, note the Orange line from D8. Its mostly software, the way I did it, I set it as output and during setup I set it to LOW, than any time I run the motor I [...]

actually I just checked some resources and they say the D8(got lucky there :)) is being pulled LOW standard, so u good to use that, as GPIO 16 is one of them weird ones, that polly noone should use.

rixmix123, 2021-11-27:
@OOIOIOO Thanks so much! Forgive me for being a noob as this is my first project working with the node mcu's and the programming of them. But do I need to modify any code with your set up?

rixmix123, 2021-11-27:
@OOIOIOO and one more question just to be clear, this set up would prevent the motor from heating up if the node MCU failed or was disconnected? Thanks again!

OOIOIOO, 2021-11-27:
> @OOIOIOO Thanks so much! Forgive me for being a noob as this is my first project working with the node mcu's and the programming of them. But do I need to modify any code with your set up?

yes in a few places where i described aboe, due to where this stuff is , it will be hard to point u exactly, as i dont know what you have. i will eventually do a pull request with my changes but im not done yet. :)

rixmix123, 2021-11-27:
Thanks again! I'll be on the lookout!

OOIOIOO, 2021-11-27:
> @OOIOIOO and one more question just to be clear, this set up would prevent the motor from heating up if the node MCU failed or was disconnected? Thanks again!

if mcu is disconnected, yes, if it failed .... hopefully, its electronics and you are playing with it, so nothing is for sure, but its better ... to the best of my knowledge. I tested it on my failed items as well and it saved it fixed it there :)

tkffaul, 2022-02-17:
@OOIOIOO just wanted to tell you how awesome of an idea this was. I modified the program and wiring like you showed and it works great! I like not having the driver powered up when the blind isnt doing anything, just made a lot of sense! I really think you should push this up as a pull request and have them modify to project!

rixmix123, 2022-02-17:
How do you modify the software? Do you make changes in Arduino code? I have the blinds up and running and connected to Wemo plugs that turn off when not in use, but I'd love another fail safe just in case! 😀

tkffaul, 2022-02-18:
Yes you have to modify the programming in Arduino... Here is how I did it... and I'm no expert, but it works...

**At the top under the var define section:**
#define STEPPER_MICROSTEP_2_PIN   12
#define STEPPER_POWER                    D8  <-- Add this Line

**At the bottom in the setup subroutine:**
void setup() {
  pinMode(STEPPER_POWER, OUTPUT);     <-- Add this Line

**Under processStepper Routine:**
  if (newPosition > currentPosition)
  {
    digitalWrite(STEPPER_POWER, HIGH);   <-- Add this line
    _Other stuff_
  
  if (newPosition < currentPosition)
  {
    digitalWrite(STEPPER_POWER, HIGH); <-- Add this line
    _Other stuff_

  if (newPosition == currentPosition && moving == true)
  {
    #if DRIVER_INVERTED_SLEEP == 1
    shadeStepper.sleepOFF();
    #endif
    #if DRIVER_INVERTED_SLEEP == 0
    shadeStepper.sleepON();
    #endif 
    digitalWrite(STEPPER_POWER, LOW);    <-- Add this Line
    _Other Stuff_

This defines STEPPER_POWER as D8 and then activates D8 as an output pin. We then flip it on before a move and shut it down when it reaches its location. Seemed simple enough, and the motor never gets warm.

rixmix123, 2022-02-18:
Awesome! Any chance you can post all the code for reference? I'll try it out!!!

petereit, 2022-05-19:
Curious, did you cut the center trace on the motor's internal PCB as Rob instructed? I was testing adding a second motor in parallel for a very large blind. I forgot to cut that center trace and it got VERY hot. The other motor I was testing had the center trace cut and always remained cool to the touch.