thehookup/Motorized_MQTT_Blinds #52: Latest version of Home Assistant doesn't show "open" or "close" after performing an action

ericullmann, 2021-03-06:
The latest version of Home Assistant utilizing this code doesn't show the status but if you revert it does. After doing some research it appears it is related to the set position command but I have tried a few things with no luck. I attached a picture as well showing that regardless of open or close the state shows unknown.
[image]

ericullmann, 2021-03-09:
I was able to get this resolved by removing the position and state topics.

christensonautohome, 2021-05-03:
this worked for me to.  not sure why but it seems the 'set_position_topic' and 'state_topic' now need to me removed from the code to show correct status.  Since I'm not too interested in figuring out the reason (only to get it working again) I'm removing these from my code and moving on.