thehookup/Motorized_MQTT_Blinds #54: Timeout Error and CheckIn Reconnected instead of OK

MileHighRy, 2021-04-25:
Set up the first one, no issues, works great.  I copied the same Arduino to the second setup but changed the MQTT client name.  It connects, shows on MQTT Explorer but the checkIn shows Reconnected instead of Ok and in my MQTT Broker log it shows the following error:

1619321299: Client LRMBlindsMCU has exceeded timeout, disconnecting.
1619321299: New connection from 10.0.0.210 on port 1883.
1619321299: New client connected from 10.0.0.210 as LRMBlindsMCU (p2, c1, k15, u'mqtthome').
1619321322: Client LRMBlindsMCU has exceeded timeout, disconnecting.

It just repeats this over and over.  I have set the initial values and sometimes can get it to move but otherwise it doesn't work from the HA Lovelace cover button or from raw values to positionCommand from MQTT Explorer.  Not sure if this is a configuration issue or something with my MQTT. Any help is greatly appreciated.

craigdthom, 2021-04-26:
I had a very similar issue ... trouble was isolated to my WiFi coverage as the signal strength fluctuated during the day when there was more activity in the house. I noticed this behavior was much less in the evening and over night. I swapped a couple of controllers and the trouble stayed with the controllers farthest away. Try swapping the controllers to see if the behavior changes.
I added a 2nd wifi router upstairs ... that solved my issue

https://imgur.com/8FNR8mx

craigdthom, 2021-04-26:
> 
> 
> I had a very similar issue ... trouble was isolated to my WiFi coverage as the signal strength fluctuated during the day when there was more activity in the house. I noticed this behavior was much less in the evening and over night. I swapped a couple of controllers and the trouble stayed with the controllers farthest away from the router. Try swapping the controllers to see if the behavior changes.
> I added a 2nd wifi router upstairs ... that solved my issue
> 
> https://imgur.com/8FNR8mx

Just another note ... I have noticed different behaviors of the NodeMCUs

1) When power to the NodeMCU is lost .... some of the NodeMCUs reset( Change state - Open/Close) and others do not
2) When my Home assistance server is restarted  .... some of the NodeMCUs reset( Change state - Open/Close) and others do not
3) When WiFi coverage changes or drop(Mesh WiFi was a headache) ... some of the NodeMCUs reset( Change state - Open/Close) and others do not

I 'm very stable now with just the two routers - 1 upstairs and one downstairs ( I  am controlling 25 blinds without any issues ) 

-Craig