thehookup/Motorized_MQTT_Blinds #57: Multiple Motors and 1 MCU Wiring ?

jeffdadamo, 2021-07-25:
I’m trying to recreate exactly what you’ve done in the video, but I have no clue how to structure the wiring for 3 stepper motors and connect them all to one MCU. Do you have a wiring schematic diagram that shows all three motors wired up?

hdubel94, 2021-07-26:
If you want all three blinds to act as one component, you should wire them in parallel off the same output. Pardon the potato image, I drafted it up quickly.

[image]

I highlighted the connection nodes in red. Whether there are 2 nodes or n nodes, they can all be combined into one theoretical node when in parallel. Think of a hub and spoke where the node is the hub and the drivers & motors are the spokes of the wheel.

I actually have 3 of these blinds running as "one" blind. As long as they all come off of the same NodeMCU outputs, they should be fine. Each wire from the output needs to go to a respective motor driver.

If you need anything clarified or have more questions please let me know and I'll try to be of help.

jeffdadamo, 2021-07-27:
[image]

So here’s a diagram I drew up based on your comments. Is this how you’d do it? Sorry for the colors, I didn’t have much to work with! 

Basically only one cat 5 coming from the Power/Buck/MCU that gets spliced to feed each of the Stepper Drivers.

hdubel94, 2021-07-27:
Yes, your diagram looks correct to me. Sorry I didn't have the time to make a more detailed one.

Your drawn solution is exactly what I have done with mine. One cat cable from the signal source (Power/Buck/MCU) that runs to each motor driver and is spliced exactly as you said/drawn.

I actually bought off Amazon a white, flat cat 6 cable and was able to feed it beneath the floor trim around the room. Not sure I'd recommend working with the flat cables unless you're very comfortable working with small wires. They were a bit of a pain to strip for the DuPont connectors.

Hope this helps and looking forward to your finished implementation!

jeffdadamo, 2021-07-27:
Awesome, thank you! Now I just need to order all the parts!

closed 2021-07-27 by jeffdadamo