thehookup/Motorized_MQTT_Blinds #58: My own Issue? Or a feature request - ESP32?

PzInTX, 2021-08-01:
Bottom line - I'd really like to do this with ESP32, rather than ESP8266. Why, because I have them and are familiar with them, but only in use with ESPHome... So I'd really love to see addition files, or maybe even a fork of this project done with ESP32 and I am more than happy to help with this!

I've been going over everything for a while now, but honestly I really struggle with Arduino IDE and coding C++ in general. I simply cannot get my code to compile. My current issue: I get a build error that "'void reconnect()' redeclared as a different kind of symbol".  The error text would indicate that the function reconnect() is used in both the PubSubClient lib as well as the ESP32WifiManager lib.

I've removed the ESP3266WiFi and ESP8266mDNS libs as they seem to conflict when I have my ESP32 board selected as the target board - hence using the ESP32WifiManager lib.

PzInTX, 2021-08-03:
After some fiddling I have this working, at least on my breadboard and desk. If anyone is interested in what I've discovered please let me know.

Although I think most people doing this probably don't struggle like I do...?

alepelc, 2025-02-14:
Do you mind sharing the ESP32 changes ?