thehookup/Motorized_MQTT_Blinds #62: Motor spazzing out when sending up/down commands

jeffdadamo1, 2021-12-22:
Hey everyone, finally got around to working on this project. I've gotten one all wired up to test before I put all three motors in line for my end setup.

However, with just the one wired up according to the wiring diagram, when I push "up" or "down" in HA, the motor "shakes" or moves a little in one direction, then back the other. The movements seem to be entirely random.

I've double and triple checked the wiring setup and I can confirm that I've done everything exactly as described. Any ideas as to what might be going on here? 

Thanks!

dogtreatfairy, 2022-01-24:
I just had this problem yesterday after scratching off the middle trace he tells you to scratch through. Try re-bridging that connection. I ended up not scratching through that trace on all of my blinds and it fixed the problem.

jeffdadamo1, 2022-01-24:
Okay, I can give that a try. I thought that was required to turn it into the type of motor we wanted/needed for this application and the code we have ? Does this mess anything else up from a power perspective ?

jeffdadamo1, 2022-01-24:
This is amazing news. I've got everything else ready to go for this project (custom printed holders for my blinds, the wiring, everything) except for this darn issue. I'll try this out and see how it goes. Thanks for the response!

closed 2022-01-24 by jeffdadamo1