thehookup/Motorized_MQTT_Blinds #66: Motor (Jaggedly?) Turns Same Way Unless Force Applied

BzowK, 2022-06-23:
Hey All - 

Had most parts sitting around and wasn't happy with current solution so gave it a shot.  Only main differences was that instead of using a buck converter (couldn't find one), I used a step up converter to step the 5V to 12V plus am using a Wemos D1 Mini instead of ESP8266.  Built it, connects with MQTT, and can control with HA.  Unfortunately, am having an issue...

**Issue**
Triggering OPEN or CLOSE moves the motor in the same direction in short bursts.  It doesn't have much torque and applying pressure to the motor while turning makes it reverse direction oddly enough.

**What I've Tried**
- Verifying wiring
- Re-bridging the 5V trace I cut on the motor's PCB (No change)
- Routing 5V to the two connected pins on the driver PCB

5V source is 2.4A and step up converter can pass 2A so would think power isn't issue but may try digging up 12V adapter and testing with it directly to driver PCB.

Any suggestions?  Thanks

reinervl, 2022-08-06:
I had the same happen to me until I realized I fried one of the stepper coils. I swapped out for a new stepper motor, redid my wiring according to [this post](https://github.com/thehookup/Motorized_MQTT_Blinds/issues/40#issuecomment-1156792619) because it seems the driver pin mappings have been updated since 2019, and bob's yur uncle! To check if you have fried coil, measure the resistance on both coils. They should be the same. If one of them reads different you have a fried coil and it's ready for the trashcan.

BzowK, 2022-08-10:
Thanks.  I tried a new motor, but got the same results.  Odd  Will test + give another shot soon and post results.  Thanks!