thehookup/Motorized_MQTT_Blinds #71: Configuration Yaml Code

nofl3dp, 2022-11-11:
I've been running the original code forever without any problems and apparently, it's now deprecated and wouldn't work.  I tried making the changes that have been suggested but it fails at line 1 when trying to restart.  I get the green check indicating everything is fine, but it won't load the file.  Here's what I have so far:

mqtt:
  cover:
     - name:  "Motorized Blinds"
       command_topic:  "BlindsMCU/blindsCommand"
       set_position_topic:  "blindsMCU/positionCommand"
       position_topic:  "blindsMCU/positionState"
       retain:  True
       payload_open:  "OPEN"
       payload_close:  "CLOSE"
       position_open:  0
       position_closed:  12
       name:  "Livingroom Motorized Blinds"
       command_topic:  "BlindsMCu_LivingRoom/blindsCommand"
       set_position_topic:  "BlindsMCU_LivingRoom/positionCommand"
       position_topic:  "BlindsMCU_Livingroom/positionState"
       retain:  True
       payload_open:  "OPEN"
       payload_close:  "CLOSE"
       position_open:  0
       position_closed:  12
  default_config:
  -tts:
      - platform:  google_translate
automation:  !include automations.yaml
script:  !include scripts.yaml
scene:  !include scene.yaml

Any help would be appreciated, thank you.

softtechs1, 2022-11-11:
I'm on HA Core 2022.11.1 with no issues. Have you checked your Mqtt server? Also need more info on green checks and files not loading etc. Can you post screenshot of what's going on?

nofl3dp, 2022-11-11:
I'm on 2022.11.1 and just tried upgrading to 2022.11.2 and now I'm getting a 404: Not found error.  The green check mark is in the file editor of the HA GUI.  If the code is valid it gives you a green check and if it's not then it gives you a red x in the upper right hand corner.  Here are a few screen shots that I was able to get.
![configuration yaml](https://user-images.githubusercontent.com/1180
[image]
13134/201431521-a456b1a8-86ef-4874-8970-dc0f6fd56d6d.jpg)

nofl3dp, 2022-11-11:
[image]

nofl3dp, 2022-11-16:
Anyone have ideas on what needs to be fixed in the config file?  The home assistant url still isn't working because of the config file problems.  I can edit the file using the cli but not sure what needs to be fixed after they updated the software.