thehookup/Motorized_MQTT_Blinds #72: PCB

Nexetion, 2022-11-21:
Has anyone attempted to make a PCB for this project? it would make it so much better!

Siumba, 2023-01-09:
I've tried creating one, I just ordered it and have therefore not tested it yet.
Attached zip holds the gerber files
[Gerber_PCB_BlindsMCU.zip](https://github.com/thehookup/Motorized_MQTT_Blinds/files/10372882/Gerber_PCB_BlindsMCU.zip)

[image]

Siumba, 2023-01-18:
Regarding my previous comment, that PCB works. However, I've created a new, updated version which can drive 2 motors, uses a different DC DC Converter and it has overvoltage protection by using a 100uf capacitor.
[image]
[image]
[Gerber_PCB_BlindsMCU_8_2023-01-18.zip](https://github.com/thehookup/Motorized_MQTT_Blinds/files/10450120/Gerber_PCB_BlindsMCU_8_2023-01-18.zip)