thehookup/Motorized_MQTT_Blinds #73: 6mm Square STL

pmontone83, 2023-01-23:
I am new to github so maybe this is just my mistake, but i cant seem to download the 6mm square stl there is no download link on the file.

gedeyenite, 2023-02-03:
Same issue for me as of 42023-02-03. The 5mm and 4mm have district DOWNLOAD buttons while the 6mm does not.

tThanks.

softtechs1, 2023-02-03:
Just click the edit down arrow and select "open in github.dev" copy all and past into a text editor and save as [filename].stl

gedeyenite, 2023-02-04:
Darn. I hoped taht would work but it produces an error when I try to import the resultant file in PRUSA Slicer.
[image]

softtechs1, 2023-02-04:
Worked for me with Ultimaker Cura 5.0.