thehookup/Motorized_MQTT_Blinds #75: Esphome

timknowlden, 2023-03-18:
Has anybody used this hardware and schematic with esphome as the software? I've got all the hardware and printing a mount but I would like to integrate using esphome and home assistant preferably.

abnormalend, 2023-03-20:
This was my plan, though I'm waiting on my hardware to come so I haven't started yet.  I'll publish my esphome code if I get it working.

timknowlden, 2023-04-03:
> This was my plan, though I'm waiting on my hardware to come so I haven't started yet. I'll publish my esphome code if I get it working.

I built these up with listed components, it turns out though that sadly the drivers are not supported by esphome

UhtredTheBold, 2023-11-04:
I haven't mounted this in my blinds yet but I've been testing different drivers on my bench.

I've successfully got esphome working with the tmc2008 drivers and 28byj 48 (and 12v bipolar mod) steppers using the standard A4988 component. The only code changes I had to make were inverting the sleep pin and increasing the range of the min/max range considerably. 

I haven't mounted everything in my blinds yet but I hope to do a full test next week. They have the advantage of being really quiet in comparison to the DRV8825. 

I can post more info if there's any interest.

koverly, 2023-11-07:
> I can post more info if there's any interest.

@UhtredTheBold - I'm looking at ordering the hardware to build this as well and am very interested in any info you have to share.

koverly, 2023-11-07:
It did stumble on this older reddit post that had ESPHome code that could be used as a reference.  https://www.reddit.com/r/homeassistant/comments/lw68lu/i_remade_the_hook_ups_motorized_blinds_with/

UhtredTheBold, 2023-11-07:
That's gold, good find. I do wonder if esphome should rename the a4988 component to generic_stepper, or something like that, because it works on at least 3 different stepper drivers.

koverly, 2023-11-25:
I have the parts now and am starting to experiment.  I'll post what I find

tronikos, 2024-04-14:
https://github.com/tronikos/esphome-blinds