thehookup/Motorized_MQTT_Blinds #76: Tasmota?

timknowlden, 2023-04-03:
I have wired all components as per schematic, I then tried integrating using ESPhome, however after a failed attempt I read that the DRV8825 is unsupported by ESPhome, SO that means I need to go arduino route, or as you say tasmota, but im not familiar with tasmota and how to integrate as a cover, before i go that route is the schematic the same for all components?

I want the most reliable option which is why I'd consider tasmota after your comment on it being more reliable inside the readme.

timknowlden, 2023-04-03:
It reads to me as though 
D5 = Relay_i
D6 = Relay 2
D7 = PWM1
D8 = Counter 1

D7&D8 are then connected together?
Am I correct?

UhtredTheBold, 2023-11-05:
You can use esphome, just use the A4988 component