thehookup/Motorized_MQTT_Blinds #82: 1 rotation from stepper motor needed

KoenVoortman, 2024-09-24:
Hi, noob/novice MQTT and Arduino user here. 

I have a Homebridge installation with an embedded MQTT broker (Aedes) and a Mqttthing plugin for devices.
That part is working, and I have connection with my NodeMCU and I can control the stepper motor.

I'm in the Netherlands and have Veneta 50mm horizontal/venetian blinds.
I've re-designed a housing for the stepper motor, driver, NodeMCU and power converter, that will fit in the blinds housing, see pictures below.

My problem is: the full open to closed is 1 rotation of the hexagon axis in the blinds housing.
How/where do I adjust settings in the code, so the stepper motor will only rotate 1 time (1x 360 degrees) when I send a 0<->100 command?
I've played with the settings in the Arduino code, and there are limited options in the MQTT code in Homebridge.

I hope somebody can help!

[image]
[image]

Sander1978-dev, 2025-01-30:
Hi Koen, I am currently printing it for the Veneta blinds (60mm) to see if it fits, can you share your STL please? That way I can see if it fits for me and I can try to reproduce it and fix it for you

KoenVoortman, 2025-01-30:
Hi Sander,

This is the most recent version. The holes are for "cooling" and routing wires internally.

I'm curious if you can fix it! Thanks!

[image]

[HorizontalBlindsMotorHousing_KVO_v7.stl.zip](https://github.com/user-attachments/files/18608872/HorizontalBlindsMotorHousing_KVO_v7.stl.zip)

Sander1978-dev, 2025-01-31:
Thank you, I just started printing it and will look at the code. I have also found this discussion (in Dutch) https://gathering.tweakers.net/forum/list_messages/2207606

with the corresponding github: https://github.com/Hordica/ESPHome-smart-shutter-tilt

However, that 3D model is for the old Veneta blinds and it uses microswitches to detect begin and end. That way it wont rotate further then needed. So the code and microswitches could be useful. 

I have that code and hardware connected, however no 3D model for it yet. I will try to see if I can get best from both worlds and will respond here when I got it working.

Sander1978-dev, 2025-01-31:
Hi Koen, I have looked at the code and have you tried modifying the line
`position_closed: 12` to `position_closed: 3` 

I devided it by 4 since you say it does 4 rotations. 

Your mod fits perfectly in my veneta 60mm blinds, however I have used a 5V stepper motor from the project I mentioned above instead of this project mentioning a 12V stepper motor. The 5V stepper motor doesnt have the power to turn the axis.

KoenVoortman, 2025-02-11:
Hi Sander, yes I think I’ve tried. 
Several settings actually. 
Will try again in a few weeks maybe. Kinda busy now. 

Good to hear the housing fits in your Veneta also!
I’ve gone through some versions ;) 
This was V1 through 5 😊
Made some minor adjustments after those. 

[image]

Sander1978-dev, 2025-03-28:
Hi Koen, today I finally got time to try the 12V setup. But even the 12V stepper motor is unable to turn my blinds. Only difference I see in the blinds is that you have the band where I have small wires connected to the horizontal blades. Did you remove the box insde the blinds where the manual control cables are?

KoenVoortman, 2025-03-30:
Hi, I have no problems with my 50mm ones. Just have to find time to finish 8 modules to complete all my blinds.

And yes, you need to remove the original manual control box. This is your problem I think.
I've put my modules on that spot in the base.